BinPy/BinPy #285: Fatal Error

Mushirahmed, 2015-08-14:
Dear BinPy Support Team,
I am installing the BinPy library using git clone method.
Following is the issue regarding to installation.
Cloning into 'BinPy'...
error: The requested URL returned error: 407 while accessing http://github.com/BinPy/BinPy.git/info/refs
fatal: HTTP request failed

debugger22, 2015-08-14:
I think this is GitHub who is stopping you. https://github.com/blog/809-git-dumb-http-transport-to-be-turned-off-in-90-days

Mushirahmed, 2015-08-14:
I am using latest version i.e 1.7.9.5 of git
Still it is not giving same error

debugger22, 2015-08-14:
Oh! I would suggest you to download the package from [here](https://github.com/BinPy/BinPy/archive/v0.3.1.tar.gz) and install it manually. Let me know if you face any problem.

salil93, 2015-08-14:
For time being, I suggest you to download the zip of the master branch.

Mushirahmed, 2015-08-14:
Dear Sudhanshu and Salil,
Thanks..
BinPy now successfully installed.

closed 2015-08-14 by debugger22