BinPy/BinPy #292: Getting an error in output and input

ArjunAbhijith, 2019-09-24:
I was trying the example of srlatch which didnt work.
 Error:
Traceback (most recent call last):
  File "c:/Users/Arjun/Desktop/Python/Python 18-20 -3.7/logic gates/GATES.py", line 13, in <module>
    NOR1.setOutput(con2)  # Set output for NOR gate
AttributeError: 'NOR' object has no attribute 'setOutput

OrkoHunter, 2019-09-24:
Hi! Looking into the code, it seems that the current method is `set_output`. Can you try that?

ArjunAbhijith, 2019-09-24:
its working now idk how i just reinstalled it. And it worked

ArjunAbhijith, 2019-09-24:
btw i have a question that how could we add a particular value in register after initialising it.
@OrkoHunter

OrkoHunter, 2019-09-24:
I am not familiar with the code myself. Also, the library has not been active for a long time. Look around the documentation and code and there might be something useful to you. :)

closed 2019-09-24 by OrkoHunter